hovren/crisp #2: AssertionError in calibration.py

sergiud, 2015-08-24:
I'm using the OpenCV camera model. After fixing several code issues and running crisp, I get the following error:

```
Creating video stream from /home/sergiu/Projects/test_sequences/building1.mp4
Creating gyro stream from /home/sergiu/Projects/test_sequences/building1_gyro.csv
Creating calibrator
Estimating time offset and camera to gyroscope rotation. Guessing gyro rate = 200.00
Traceback (most recent call last):
  File "parrot_ardrone_dataset.py", line 70, in <module>
    calibrator.initialize(gyro_rate=GYRO_RATE_GUESS)
  File "/home/sergiu/Projects/crisp/crisp/calibration.py", line 125, in initialize
    R = self.find_initial_rotation()
  File "/home/sergiu/Projects/crisp/crisp/calibration.py", line 290, in find_initial_rotation
    assert data.shape == (6, len(gyro_axes))
AssertionError
```

`gyro_axes` remains empty. I assume that estimating the initial rotation fails resulting in the error. Is this correct?

hovren, 2015-08-25:
What version are you using? There was a bug in v2.0 where the number of collected rotation axes would not match, which should have been fixed in v2.1.

Can you debug and check what `data.shape` actually contains?

hovren, 2015-08-25:
Also, if the "several code issues" you have found are still present in v2.1 I would be very happy to receive patches for those.
For example, as should hopefully be clear from the documentation, the OpenCV camera model has not received a lot of testing since I have never used it for my own experiments so far.

sergiud, 2015-08-25:
I used 2fd0270e2b7397d1432c3ac88f7078044be785c1. As for the code issues, I'll create pull requests later today.

hovren, 2015-08-25:
Ok, in that version the issue I talked about should have been fixed. The more problematic (and embarassing) residual bug I found and fixed in #d323508b97ba657600b2e6bdc62b97fa72e10b78 is not, however. So I would recommend updating to the latest version before starting any experiments.

Also, to know what is going on in this issue, it would still be good to know the value of `data.shape` that is triggering the assert.

sergiud, 2015-08-25:
Same issue with 2.1. As a side note, I had to change project and unproject to return (2,N) and (3,N) arrays, otherwise I'm getting `ValueError: operands could not be broadcast together with shapes (2,183) (183,1,2) `. `data.shape` is (2,0).

hovren, 2015-08-25:
The problem is, as you already noticed, that you didn't have enough axes-pairs to perform the rotation estimation (at least 2 are required). I added a check for this. Can you please check out the newest version from the master branch and test if that solves the problem?

sergiud, 2015-08-25:
It does solve my problem in the sense that I get a message that tells me about failed initialization. So, basically I can't use the video sequence with your calibration approach.

I tried another video sequence and now I get a different error:

```
Creating video stream from /home/sergiu/Projects/test_sequences/redsquare11.mp4
Creating gyro stream from /home/sergiu/Projects/test_sequences/redsquare11_gyro.csv
Creating calibrator
Estimating time offset and camera to gyroscope rotation. Guessing gyro rate = 200.00
/usr/lib/python2.7/site-packages/numpy/core/_methods.py:59: RuntimeWarning: Mean of empty slice.
  warnings.warn("Mean of empty slice.", RuntimeWarning)
/usr/lib/python2.7/site-packages/numpy/core/_methods.py:71: RuntimeWarning: invalid value encountered in true_divide
  ret = ret.dtype.type(ret / rcount)
(6, 15)
[array([ 0.14526438, -0.05357122, -0.98794149]), array([ 0.99412952, -0.01678371, -0.10688685]), array([-0.0611561 ,  0.64598077,  0.76089997]), array([-0.41998796, -0.80055735,  0.42745531]), array([-0.50649943, -0.19890679, -0.83898415]), array([ 0.07588637,  0.05336562,  0.99568739]), array([-0.6664287 , -0.74041999, -0.08747011]), array([ 0.20088705,  0.97777033, -0.06007977]), array([ 0.90378761,  0.42743605, -0.0215957 ]), array([ 0.75751925,  0.65242953,  0.02236749]), array([ 0.51498225, -0.8449579 , -0.14435866]), array([ 0.66171341, -0.34334071, -0.66652271]), array([ 0.92792801, -0.09279939,  0.36102338]), array([ 0.79558824,  0.60016459, -0.08271525]), array([-0.74115061, -0.66290071, -0.10610569])]
Running calibration. This can take a few minutes.
/home/sergiu/Projects/crisp/crisp/calibration.py:188: UserWarning: Could not use the requested 1500 tracks, since only 15 were available in the slice data.
  warnings.warn("Could not use the requested {} tracks, since only {} were available in the slice data.".format(max_tracks, available_tracks))
Traceback (most recent call last):
  File "parrot_ardrone_dataset.py", line 77, in <module>
    calibrator.calibrate()
  File "/home/sergiu/Projects/crisp/crisp/calibration.py", line 199, in calibrate
    leastsq_result = scipy.optimize.leastsq(optimization_func, x0, args=func_args, full_output=True, ftol=1e-10, xtol=1e-10, maxfev=max_eval)
  File "/usr/lib/python2.7/site-packages/scipy/optimize/minpack.py", line 371, in leastsq
    shape, dtype = _check_func('leastsq', 'func', func, x0, args, n)
  File "/usr/lib/python2.7/site-packages/scipy/optimize/minpack.py", line 20, in _check_func
    res = atleast_1d(thefunc(*((x0[:numinputs],) + args)))
  File "/home/sergiu/Projects/crisp/crisp/calibration.py", line 434, in optimization_func
    Y = camera.unproject(y)
  File "/home/sergiu/Projects/crisp/crisp/camera.py", line 325, in unproject
    undist_image_points = cv2.undistortPoints(np.concatenate(image_points).reshape(-1,1,2), self.camera_matrix, self.dist_coefs, P=self.camera_matrix)
ValueError: zero-dimensional arrays cannot be concatenated
```

`np.concatenate` was added by me, as OpenCV threw an exception without it. data.shape here is `(6, 15)`.

Did you by any chance calibrate your camera using OpenCV and can provide me the camera intrinsic matrix along the lens distortion coefficients?

hovren, 2015-08-25:
No, for my experiments I used the `AtanCameraModel`. The `OpenCVCameraModel` was added since I realize that most people would probably expect to use that. I thought I had mentioned somewhere in the documentation that it has received little to no testing. Seems like I didn't. Sorry about that.
The camera parameters for that camera is available in the example dataset and script as mentioned in the README.

I found an error in `OpenCVCameraModel.project` which had it return the wrong shape. That is why you had to add the `np.concatenate` call. This should no longer be necessary.

Unless your video is extremely challenging, it should be able to find at least some slices with a valid rotation. But since the camera projections were wrong, this is probably why it failed.

If 145d228505ef005465032931406c3b80ce5853b0 didn't fix it, would you mind sharing the video and gyro data with me for testing purposes? My e-mail is in the README if you don't feel like disclosing it publicly.

sergiud, 2015-08-25:
Now I get:

```
Creating video stream from /home/sergiu/Projects/gopro-gyro-dataset/redsquare11.mp4
Creating gyro stream from /home/sergiu/Projects/gopro-gyro-dataset/redsquare11_gyro.csv
Creating calibrator
Estimating time offset and camera to gyroscope rotation. Guessing gyro rate = 200.00
/usr/lib/python2.7/site-packages/numpy/core/_methods.py:59: RuntimeWarning: Mean of empty slice.
  warnings.warn("Mean of empty slice.", RuntimeWarning)
/usr/lib/python2.7/site-packages/numpy/core/_methods.py:71: RuntimeWarning: invalid value encountered in true_divide
  ret = ret.dtype.type(ret / rcount)
OpenCV Error: Assertion failed (src.isContinuous() && (src.depth() == CV_32F || src.depth() == CV_64F) && ((src.rows == 1 && src.channels() == 2) || src.cols*src.channels() == 2)) in undistortPoints, file /build/opencv/src/opencv-2.4.11/modules/imgproc/src/undistort.cpp, line 394
Traceback (most recent call last):
  File "parrot_ardrone_dataset.py", line 70, in <module>
    calibrator.initialize(gyro_rate=GYRO_RATE_GUESS)
  File "/home/sergiu/Projects/crisp/crisp/calibration.py", line 125, in initialize
    R = self.find_initial_rotation()
  File "/home/sergiu/Projects/crisp/crisp/calibration.py", line 257, in find_initial_rotation
    _slice.estimate_rotation(self.video.camera_model, ransac_threshold=7.0) # sets .axis and .angle memebers
  File "/home/sergiu/Projects/crisp/crisp/videoslice.py", line 50, in estimate_rotation
    do_translation=False)
  File "/home/sergiu/Projects/crisp/crisp/rotations.py", line 281, in estimate_rotation_procrustes_ransac
    X = camera.unproject(x)
  File "/home/sergiu/Projects/crisp/crisp/camera.py", line 325, in unproject
    undist_image_points = cv2.undistortPoints(image_points.T.reshape(-1,1,2), self.camera_matrix, self.dist_coefs, P=self.camera_matrix)
cv2.error: /build/opencv/src/opencv-2.4.11/modules/imgproc/src/undistort.cpp:394: error: (-215) src.isContinuous() && (src.depth() == CV_32F || src.depth() == CV_64F) && ((src.rows == 1 && src.channels() == 2) || src.cols*src.channels() == 2) in function undistortPoints
```

This is the same error I was getting without np.concatenate.

hovren, 2015-08-26:
I found another error in `OpenCVCameraModel.unproject` that is now committed.
With the current master branch I could run the calibration on the data you provided me with.
Please let me know if this fixed the problem.

sergiud, 2015-08-26:
Works now, thanks.

hovren, 2015-08-26:
Thanks for helping me find those bugs!

commit 37f93e64 (2015-08-25): Check enough available axes-pairs for rotation estimation

This fixes #2
closed 2015-08-26 by hovren