hovren/crisp #4: Fast Integrate Dependency

goldbattle, 2015-09-07:
I was trying to install this, but it errors on build that it is missing a `crisp/fastintegrate/fastintegrate.c` file. I see that in the [gitignore](https://github.com/hovren/crisp/blob/master/.gitignore#L15) there as .c file is excluded. Where am I suppose to download this file so I can compile the project?

I am on ubuntu 14.04 and here is the error:
```console
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/crisp
copying crisp/cli.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/stream.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/rotations.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/calibration.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/znccpyr.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/videoslice.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/l3g4200d.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/remove_slp.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/tracking.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/__init__.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/timesync.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/imu.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/ransac.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/camera.py -> build/lib.linux-x86_64-2.7/crisp
copying crisp/pose.py -> build/lib.linux-x86_64-2.7/crisp
running build_ext
building 'crisp.fastintegrate' extension
creating build/temp.linux-x86_64-2.7
creating build/temp.linux-x86_64-2.7/crisp
creating build/temp.linux-x86_64-2.7/crisp/fastintegrate
x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/lib/python2.7/dist-packages/numpy/core/include -I/usr/include/python2.7 -c crisp/fastintegrate/fastintegrate.c -o build/temp.linux-x86_64-2.7/crisp/fastintegrate/fastintegrate.o
x86_64-linux-gnu-gcc: error: crisp/fastintegrate/fastintegrate.c: No such file or directory
x86_64-linux-gnu-gcc: fatal error: no input files
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 4
```

hovren, 2015-09-07:
Hi!

If you have cloned the master branch of the repository you need `Cython` to build.
The C-file is generated by Cython and only distributed with the source releases, to allow installing/building without the Cython dependency.
This should probably be more clear in the documentation. Fortunately, installing Cython should be pretty straightforward either from pip, conda or from the Ubuntu repositories, and after that building should work as intended.

goldbattle, 2015-09-07:
Perfect!
It seems to be running now.
Installed using pip: `pip install cython`

closed 2015-09-07 by goldbattle