hovren/crisp #6: zncc: Division by zero

nesnes, 2016-05-27:
Hi, I have some troubles using crisp with my own samples. I took three sets of qvga@30fps video with 30Hz gyroscope and one set of 720p@60fps and 100Hz gyroscope and every one of those is giving me a Division by Zero error in the calibration.initialize(...) . I've tried to tune my setting values in your gopro_dataset_example.py, but the error is still there...

```
➜  examples python gopro_dataset_example.py dataset/samples/Set1/test.avi
Creating video stream from dataset/samples/Set1/test.avi
Creating gyro stream from dataset/samples/Set1/test_gyro.csv
Post processing gyroscope data to remove frequency spike noise
Creating calibrator
Estimating time offset and camera to gyroscope rotation. Guessing gyro rate = 30.00
/usr/lib/python2.7/dist-packages/numpy/core/_methods.py:55: RuntimeWarning: Mean of empty slice.
  warnings.warn("Mean of empty slice.", RuntimeWarning)
/usr/lib/python2.7/dist-packages/numpy/core/_methods.py:67: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Traceback (most recent call last):
  File "gopro_dataset_example.py", line 96, in <module>
    calibrator.initialize(gyro_rate=GYRO_RATE_GUESS)
  File "build/bdist.linux-x86_64/egg/crisp/calibration.py", line 128, in initialize
  File "build/bdist.linux-x86_64/egg/crisp/calibration.py", line 230, in find_initial_offset
  File "build/bdist.linux-x86_64/egg/crisp/timesync.py", line 107, in sync_camera_gyro
  File "build/bdist.linux-x86_64/egg/crisp/znccpyr.py", line 273, in find_shift_pyr
  File "build/bdist.linux-x86_64/egg/crisp/znccpyr.py", line 175, in zncc
ZeroDivisionError: division by zero
```

The debug output is saying : 
```
DEBUG:crisp:Number of slices: 84
DEBUG:crisp:Generating optical flow magnitude. This can take minutes depending on video length
DEBUG:root:pyramid size = 7
DEBUG:root:size of first element 2243 
DEBUG:root:size of last element 35 
```

Do you have any idea of what I can be doing wrong?

hovren, 2016-08-22:
Hi, I am terribly sorry for the super late reply!

I have run into this as well, and will try to get it fixed as soon as possible.
Sorry for the inconvenience!

ShreyasManjunath, 2020-11-23:
Is the fix available? I still encounter the bug.

hovren, 2020-11-25:
Last time I checked I think it might have been a problem where an input array shape was wrong. If you are using your own data make sure that it is in the right format (e.g. the gyro samples).

The problem should be connected to the warning message about taking the mean of an empty slice. Please try to see why that happens.