hovren/crisp #9: Video Stabilization Method

yang778, 2016-11-01:
Hi, @hovren 
Thank you for your excellent work. After calibration, I need to stabilize the video. But I'm not sure which method to use. I want to stabilize the video on a UAV in real time. Do you have some recommendations?
Thank you in advance.

hovren, 2016-11-01:
Hi!
Unfortunately the actual stabilization is out of scope for this software and I don't have any recommendations on methods that are realtime suitable.
Good luck with your search!

closed 2016-11-01 by hovren

naldoxable, 2020-09-16:
Hi, what about non realtime stabilization? I have seen this [Video](http://www.cvl.isy.liu.se/research/datasets/gopro-gyro-dataset/ICRA15_1759_VI_i.mp4) and it looks amazing! How did you managed to get this result? Do you have some code or repository to advice? Thanks

hovren, 2020-11-25:
> Hi, what about non realtime stabilization? I have seen this [Video](http://www.cvl.isy.liu.se/research/datasets/gopro-gyro-dataset/ICRA15_1759_VI_i.mp4) and it looks amazing! How did you managed to get this result? Do you have some code or repository to advice? Thanks

Hi @naldoxable ,

Everything about the method is described in the paper that is referenced in the README. This repository contains the code for the Camera-gyro synchronization. The video stabilization code is however not available, but it is described in the paper.

terencetay, 2021-09-18:
Hello, like others here, I'm interested in reproducing your video stabilization results. As you suggested, I read your ICRA 2015 paper where you said that you used the method described in the paper below for stabilization. But the paper below uses feature point tracking with KLT to estimate the camera rotation. It does not use a gyroscope. Can you provide more details about how you used that method with your gyroscope calibration to stabilize the videos you showed in your paper? Thank you.

E. Ringaby and P.-E. Forss´en, “Efficient video rectification and stabilisation
for cell-phones,” International Journal of Computer Vision,
vol. 96, no. 3, pp. 335–352, February 2012.

hovren, 2021-09-18:
Hi,

No I can't really give much more info than what is in the paper: instead of
computing the rotation from features we use gyroscopic measurements. When
the rotation sequence has been found we apply the method from Ringaby et al.

terencetay, 2021-09-18:
Thank you. It is useful to understand how your implementation is different from what was described by Ringaby et al.