hovren/crisp #12: AttributeError: 'numpy.ndarray' object has no attribute 'sqrt'

roschachJoke, 2019-03-12:
Hi, when I run gopro_dataset_example.py with sample video in gopro-gyro-dataset， I got following errors in calibration.py in line:

rn = np.sqrt((X[0,:] - wx)**2 + (X[1,:] - wy)**2)

in function apply.

All three video: rotation.MP4, walk.MP4, rccar.MP4 will fall into this error after some successfully computation for previous calling.

I dumped the X data and found that X will be like below when reporting this error.
print(X) -->
[[array([-0.44746115])]
 [array([0.02795225])]
 [array([1.])]]

Could you help me found out why?
Thanks a lot!

roschachJoke, 2019-03-12:
The error is: AttributeError: 'numpy.ndarray' object has no attribute 'sqrt'

yf13, 2019-05-05:
I met the same issue with spicy 1.2.1 and numpy 1.16.3

I resolved the issue by forcing the input array X to have "float" dtype:

```
        X = points if not points.ndim == 1 else points.reshape((points.size, 1))
        
        if X.dtype != np.dtype('float'):
            X = X.astype('float')

        wx, wy = self.wc
```

Then it can move ahead. A similar fix to the GoPro_example.py finally lead to the following results:

```
$ ./gopro_example.py data/walk.MP4 
Creating video stream from data/walk.MP4
Creating gyro stream from data/walk_gyro.csv
Post processing L3G4200D gyroscope data to remove frequency spike noise
Creating calibrator
Estimating time offset and camera to gyroscope rotation. Guessing gyro rate = 853.86
No handlers could be found for logger "crisp.videoslice"
Running calibration. This can take a few minutes.
Parameters
--------------------
    gyro_rate = 854.494579007
  time_offset = 3.39245322817
      gbias_x = -0.00395476658282
      gbias_y = -0.0116545311793
      gbias_z = 0.0108619096344
        rot_x = -0.00784905963239
        rot_y = 0.00497907927654
        rot_z = -1.53902067358

Compare with reference data

   Parameter     Reference    Optimized    Difference 
      gyro_rate  8.547640E+02  8.544946E+02  2.693984E-01
    time_offset  3.385811E+00  3.392453E+00  -6.641780E-03
        gbias_x  -3.765542E-03  -3.954767E-03  1.892246E-04
        gbias_y  -1.137625E-02  -1.165453E-02  2.782788E-04
        gbias_z  1.058667E-02  1.086191E-02  -2.752380E-04
          rot_x  -1.485679E-02  -7.849060E-03  -7.007731E-03
          rot_y  -1.081410E-02  4.979079E-03  -1.579318E-02
          rot_z  -1.549487E+00  -1.539021E+00  -1.046595E-02
Reference rotation
[[ 0.02128913  0.99977046  0.00240653]
 [-0.99963949  0.02124682  0.01641561]
 [ 0.01636071 -0.00275514  0.99986236]]
Optimized rotation
[[ 0.03176743  0.99946188  0.0081715 ]
 [-0.99949383  0.03175239  0.00196495]
 [ 0.00170443 -0.00822979  0.99996468]]
Angle difference: 1.0774 degrees
```

Actually the result doesn't looks like the same across multiple runs. Sometimes the angle difference is 1. deg, sometimes it was -0.5deg.

ouyangandy, 2019-06-24:
it is not work by add
 " if X.dtype != np.dtype('float'):
            X = X.astype('float')"
how to fix this?AttributeError: 'numpy.ndarray' object has no attribute 'sqrt'

ouyangandy, 2019-06-24:
> Hi, when I run gopro_dataset_example.py with sample video in gopro-gyro-dataset， I got following errors in camera.py in line:
> 
> ```
> rn = np.sqrt((X[0,:] - wx)**2 + (X[1,:] - wy)**2)
> ```
> 
> AttributeError: 'numpy.ndarray' object has no attribute 'sqrt'
> 
> in function apply.
> 
> All three video: rotation.MP4, walk.MP4, rccar.MP4 will fall into this error after some successfully computation for previous calling.
> 
> I dumped the X data and found that X will be like below when reporting this error.
> print(X) -->
> [[array([-0.44746115])]
> [array([0.02795225])]
> [array([1.])]]
> 
> Could you help me found out why?
> Thanks a lot!

it is not work by add
" if X.dtype != np.dtype('float'):
X = X.astype('float')"
how to fix this?AttributeError: 'numpy.ndarray' object has no attribute 'sqrt'

naldoxable, 2019-08-31:
```
python3 gopro_dataset_example.py ~/Scrivania/rccar.MP4 
Creating video stream from /home/naldoxable/Scrivania/rccar.MP4
Creating gyro stream from /home/naldoxable/Scrivania/rccar_gyro.csv
Post processing L3G4200D gyroscope data to remove frequency spike noise
Creating calibrator
Estimating time offset and camera to gyroscope rotation. Guessing gyro rate = 853.86
Running calibration. This can take a few minutes.
Traceback (most recent call last):
  File "gopro_dataset_example.py", line 82, in <module>
    calibrator.calibrate()
  File "/home/naldoxable/Scrivania/crisp/build/lib.linux-x86_64-3.7/crisp/calibration.py", line 199, in calibrate
    leastsq_result = scipy.optimize.leastsq(optimization_func, x0, args=func_args, full_output=True, ftol=1e-10, xtol=1e-10, maxfev=max_eval)
  File "/usr/lib/python3/dist-packages/scipy/optimize/minpack.py", line 384, in leastsq
    shape, dtype = _check_func('leastsq', 'func', func, x0, args, n)
  File "/usr/lib/python3/dist-packages/scipy/optimize/minpack.py", line 26, in _check_func
    res = atleast_1d(thefunc(*((x0[:numinputs],) + args)))
  File "/home/naldoxable/Scrivania/crisp/build/lib.linux-x86_64-3.7/crisp/calibration.py", line 446, in optimization_func
    xhat = camera.project(Xhat)
  File "/home/naldoxable/Scrivania/crisp/build/lib.linux-x86_64-3.7/crisp/camera.py", line 237, in project
    X = self.apply(XU)
  File "/home/naldoxable/Scrivania/crisp/build/lib.linux-x86_64-3.7/crisp/camera.py", line 207, in apply
    rn = np.sqrt((X[0,:] - wx)**2 + (X[1,:] - wy)**2)
AttributeError: 'numpy.ndarray' object has no attribute 'sqrt'
```

#### that is what i get.
Compiled and installed and run with python3 but using older version dosn't resolve

naldoxable, 2019-08-31:
hope you can fix that, i need this in combination with the gpmf extractor to defeat reelsteady go :)